3DMIN/3DMIN.github.io #1: <span class="note"> für note boxes?

LFSaw, 2016-03-19:
```
<span class="note">for the following steps it is recommended to work in pairs of two</span>
```

[image]

pppws, 2016-03-20:
(bleibt man hier englisch?)
komme erst morgen vormittag dazu. denke bis mittag ist das bis auf den firefix-bug drin!

pppws, 2016-03-21:
die auszeichnungen sind wie besprochen drin.

## h2
**bold** – fett, auszeichnung im fließtext
*em* – kursiv, fließtext und/oder bildunterschriften
<span class="note"></span> – weißer rahmen um text, hinweis
<pre></pre> – code (noch ohne farbige codierung)

closed 2016-03-21 by pppws

LFSaw, 2016-03-21:
danke!
englisch: ist recht egal... ja für uns nur, irgendwie. Ich schreibe solche sachen eher in englisch, hoffe das ist ok.